Ohm-Slah/Gallagher-Station-Duke-Energy-Museum-Exhibit #66: Add endpoints for all drawings

FlyingBowser, 2022-02-28:
add endpoints for all board going to an external connector (motors or 7segment LED for example)

Ohm-Slah, 2022-03-10:
@FlyingBowser Where are you at with this?

FlyingBowser, 2022-03-10:
I am going to do 1100 endpoints today then do one or two sets per day to commit more time to details on different sets

Ohm-Slah, 2022-03-10:
Sounds good. Fyi, I have moved all BOM material to a centralized file to help with simplicity. The commit can be found here: [456dcd23a9117a93bf4793d457157b3c41bb55c1](https://github.com/Ohm-Slah/Gallagher-Station-Duke-Energy-Museum-Exhibit/commit/456dcd23a9117a93bf4793d457157b3c41bb55c1)
Thanks

FlyingBowser, 2022-03-13:
1100 and 1200 done, need reviewed

FlyingBowser, 2022-03-14:
All current drawings reviewed, currently working on 1310 and 1410

FlyingBowser, 2022-03-14:
Update: 1310 not needed